BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #4: Compile issues

happytm, 2020-06-09:
Looks very useful code. I am having issues compiling it. Following are some error messages at the end :

Arduino: 1.8.9 (Windows 7), Board: "LOLIN D32, Default, 80MHz, 921600, None"

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\file_system.h: In function 'bool __writeEntireFileWithoutSemaphore__(String, char*)':

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\file_system.h:114:3: error: control reaches end of non-void function [-Werror=return-type]

   }  

   ^

In file included from C:\Users\WinCycle\Documents\Arduino\Esp32_web_ftp_telnet_server_template\Esp32_web_ftp_telnet_server_template.ino:233:0:

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\telnetserver.hpp: In static member function 'static bool telnetServer::__pingRecv__(telnetServer::__pingDataStructure__*, TcpConnection*, int)':

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\telnetserver.hpp:883:7: error: control reaches end of non-void function [-Werror=return-type]

       }  

       ^

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\telnetserver.hpp: In static member function 'static String telnetServer::__curl__(TcpConnection*, String, String)':

c:\users\wincycle\appdata\local\temp\arduino_build_857363\sketch\servers\telnetserver.hpp:1064:7: error: control reaches end of non-void function [-Werror=return-type]

       }

       ^

cc1plus.exe: some warnings being treated as errors

Using library WiFi at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library SPIFFS at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS 
Using library FS at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
exit status 1
format '%d' expects argument of type 'int', but argument 3 has type 'long unsigned int' [-Werror=format=]

BojanJurca, 2020-06-09:
Compiler messages look a little strange. I compile the code with Arduino without problem. I found a topic that may explain what is wrong:

https://stackoverflow.com/questions/6171500/what-does-control-reaches-end-of-non-void-function-mean

I thing putting return <whatever> at the end of the functions may solve your problem:

  bool __writeEntireFileWithoutSemaphore__ (String fileContent, char *fileName) {
    File file;
    if ((bool) (file = SPIFFS.open (fileName, "w")) && !file.isDirectory ()) {
      if (file.printf (fileContent.c_str ()) != strlen (fileContent.c_str ())) { // can't write file
        file.close ();
        Serial.printf ("[%10d] [file_system] can't write %s\n", millis (), fileName);
        return false;                
      } else {
        file.close ();
        return true;        
      }
    }
	**return false; // never executes**
  }

BojanJurca, 2020-06-09:
I just uploaded latest files into “servers” folder, adding a few “return” changes just in case. Please, use latest Arduino 1.8.12 for compiling. It should work.

happytm, 2020-06-09:
Thank you but it is still giving same error messages below:

c:\users\wincycle\appdata\local\temp\arduino_build_988697\sketch\servers\telnetserver.hpp: In function 'void __iw__(TcpConnection*)':

c:\users\wincycle\appdata\local\temp\arduino_build_988697\sketch\servers\telnetserver.hpp:1160:100: error: array subscript is above array bounds [-Werror=array-bounds]

             s += String (netif->name [0]) + String (netif->name [1]) + String ((int) netif->name [2]) + "     hostname: " + (netif->hostname ? String (netif->hostname) : "") + "\r\n" +

                                                                                                    ^

cc1plus.exe: some warnings being treated as errors

Using library WiFi at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library SPIFFS at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPIFFS 
Using library FS at version 1.0 in folder: C:\Users\WinCycle\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
exit status 1
suggest parentheses around assignment used as truth value [-Werror=parentheses]

Thanks.

happytm, 2020-06-09:
There are toomany compile errors so I am attaching Error log file here:

[errorlog.txt](https://github.com/BojanJurca/Esp32_web_ftp_telnet_server_template/files/4754429/errorlog.txt)

Thanks.

BojanJurca, 2020-06-09:
I'll try to help you soon. Somehow I managed to completely ruin my Arduino installation trying that and can't get it to work.

happytm, 2020-06-09:
Sorry to cause harm.

Thanks.

BojanJurca, 2020-06-11:
I managed to set up Arduino again and I'm working on compiler warnings. I can not go through each and every setting though. So, which time zone are you in?

happytm, 2020-06-11:
EST.New York.

BojanJurca, 2020-06-11:
I've got rid of all compiler warnings but 4 so far, please bear with them, at least for a while. I added a lot of brackets and type casting and found a few bugs meanwhile. Perhaps it is not a bad idea to set Compiler warnings to "All" after all. I’m not 100% sure that everything still works as it did. If you notice something, please let me know.

BojanJurca, 2020-06-12:
By the way, I haven't packed everything into a new release yet so, please, just download the latest files and compile them. I'll wait with a new release for a while if some bugs pop up.

happytm, 2020-06-12:
Thank you for your efforts. Now it compiles but go in boot loop with following messages:

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[        37] [ESP32] wakeup was not caused by deep sleep: 0.
[        85] [file system] SPIFFS mounted.
[        85] [SPIFFS] mounted.
[       133] [file_system] can't read /etc/passwd
[       133] [user_management] /etc/passwd does not exist or it is empty, creating a new one ... [       362] [file_system] can't write /etc/passwd
error creating /etc/passwd.
[       410] [file_system] can't read /etc/shadow
[       410] [user_management] /etc/shadow does not exist or it is empty, creating a new one ... [       642] [file_system] can't write /etc/shadow
error creating /etc/shadow.
[       690] [file_system] can't read /network/interfaces
[       690] [network] /network/interfaces does not exist, creating new one.
[       905] [file_system] can't write /network/interfaces
[       905] Unable to create file /network/interfaces.

[       952] [file_system] can't read /etc/wpa_supplicant.conf
[       952] [network] /etc/wpa_supplicant.conf does noes exist, creating new one.
[      1182] [file_system] can't write /etc/wpa_supplicant.conf
[      1182] Unable to create file /etc/wpa_supplicant.conf.
[      1230] [file_system] can't read /etc/dhcpcd.conf
[      1230] [network] /etc/dhcpcd.conf does noes exist, creating new one.
[      1444] [file_system] can't write /etc/dhcpcd.conf
[      1444] Unable to create file /etc/dhcpcd.conf.

[      1492] [file_system] can't read /etc/hostapd/hostapd.conf
[      1492] [network] /etc/hostapd/hostapd.conf does noes exist, creating new one.
[      1723] [file_system] can't write /etc/hostapd/hostapd.conf
[      1723] Unable to create file /etc/hostapd/hostapd.conf.

[      1770] [file system] can't read /etc/wpa_supplicant.conf
[      1818] [file system] can't read /network/interfaces
[      1865] [file system] can't read /etc/hostapd/hostapd.conf
[      1913] [file system] can't read /etc/dhcpcd.conf
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088c55 on core 0

Backtrace: 0x4008c820:0x3ffb2660 0x4008ca51:0x3ffb2680 0x40088c55:0x3ffb26a0 0x4014a79e:0x3ffb26e0 0x4014aa7a:0x3ffb2700 0x4013912c:0x3ffb2720 0x40139195:0x3ffb2740 0x401349b2:0x3ffb2760 0x40133420:0x3ffb2780 0x40137e54:0x3ffb27c0 0x400d2b0b:0x3ffb27e0 0x40088f69:0x3ffb2840

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
[        37] [ESP32] wakeup was not caused by deep sleep: 0.
[        85] [file system] SPIFFS mounted.
[        86] [SPIFFS] mounted.
[       133] [file_system] can't read /etc/passwd
[       133] [user_management] /etc/passwd does not exist or it is empty, creating a new one ... [       366] [file_system] can't write /etc/passwd
error creating /etc/passwd.
[       413] [file_system] can't read /etc/shadow
[       413] [user_management] /etc/shadow does not exist or it is empty, creating a new one ... [       646] [file_system] can't write /etc/shadow
error creating /etc/shadow.
[       694] [file_system] can't read /network/interfaces
[       694] [network] /network/interfaces does not exist, creating new one.
[       909] [file_system] can't write /network/interfaces
[       909] Unable to create file /network/interfaces.

[       956] [file_system] can't read /etc/wpa_supplicant.conf
[       956] [network] /etc/wpa_supplicant.conf does noes exist, creating new one.
[      1189] [file_system] can't write /etc/wpa_supplicant.conf
[      1189] Unable to create file /etc/wpa_supplicant.conf.
[      1236] [file_system] can't read /etc/dhcpcd.conf
[      1237] [network] /etc/dhcpcd.conf does noes exist, creating new one.
[      1453] [file_system] can't write /etc/dhcpcd.conf
[      1453] Unable to create file /etc/dhcpcd.conf.

[      1500] [file_system] can't read /etc/hostapd/hostapd.conf
[      1500] [network] /etc/hostapd/hostapd.conf does noes exist, creating new one.
[      1730] [file_system] can't write /etc/hostapd/hostapd.conf
[      1731] Unable to create file /etc/hostapd/hostapd.conf.

[      1778] [file system] can't read /etc/wpa_supplicant.conf
[      1825] [file system] can't read /network/interfaces
[      1873] [file system] can't read /etc/hostapd/hostapd.conf
[      1920] [file system] can't read /etc/dhcpcd.conf
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40088c55 on core 0

Backtrace: 0x4008c820:0x3ffb2660 0x4008ca51:0x3ffb2680 0x40088c55:0x3ffb26a0 0x4014a79e:0x3ffb26e0 0x4014aa7a:0x3ffb2700 0x4013912c:0x3ffb2720 0x40139195:0x3ffb2740 0x401349b2:0x3ffb2760 0x40133420:0x3ffb2780 0x40137e54:0x3ffb27c0 0x400d2b0b:0x3ffb27e0 0x40088f69:0x3ffb2840

Rebooting...
ets Jun  8 2016 00:22:57
`

BojanJurca, 2020-06-12:
At first glance it looks like board dependent problem. I have tested sketch on a few different boards but unfortunately, I do not have LOLIN D32 at my disposal. We can try a few things if you are willing to do so but it would probably require a few iterations.
What I can see from the log is that there is a problem with file system. Let’s try to solve this first. Configuration files are stored on the file system and hence nothing else can work. I can see that sketch is unsuccessfully trying to write configuration files on SPIFFS. 
I suggest the following:

1.
Check Tools | Partition scheme if any partition scheme with SPIFFS is selected.

2.
Uncomment the following line in void setup () function:
// SPIFFS.format (); 
then compile and run sketch only 1 time. This will force formatting of built-in flash disk with SPIFFS file system. After this, creation of configuration files should succeed. (please see Setup instruction prior to doing that – you have to change default settings to match your needs before configuration files are created. At least you have to change #define DEFAULT_STA_SSID  "YOUR-STA-SSID" and  #define DEFAULT_STA_PASSWORD  "YOUR-STA-PASSWORD" in network.h with your WiFi settings)
3.
Comment 
// SPIFFS.format (); 
again. You don’t want to format flash disk each time ESP32 restarts. Compile and run the sketch again.

There is a chance that this will do the trick. Either ways please let me know.

happytm, 2020-06-13:
Finally after spiffs format as you suggested everything works fine. It was worth the patience and your efforts. Now it is play time. There is lot to explore.

Thanks.

BojanJurca, 2020-06-13:
I’m glad I could help. You are welcome. I see that I must improve a user experience of the first use.

Have a nice day.

closed 2020-06-13 by BojanJurca