BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #6: Are there any plans to implement mime types for web servers?

tomoyan596, 2021-02-26:
Hello 😊

It's not a problem, but png, jpg, mp4 ... all are Content-Type: text / html ;.
It is useful to have a mime-types.h definition.

Thank you for the wonderful template 🙇

BojanJurca, 2021-02-28:
Thank you for your feedback. I'll keep it in mind for the next release which will happen, well, when I have time. 

Meanwhile you can use a workaround by adding the following code (or something similar) into httpRequestHandler function:

String httpRequestHandler (String& httpRequest, httpServer::wwwSessionParameters *wsp) { // - has to be reentrant!

  if (httpRequest.indexOf (".png ") >= 0)       wsp->setHttpResponseHeaderField ("Content-Type", "image/png");
  else if (httpRequest.indexOf (".html ") >= 0) wsp->setHttpResponseHeaderField ("Content-Type", "text/html");
  else                                          wsp->setHttpResponseHeaderField ("Content-Type", "text/plain");
  // ...

You can also put it into httpServer.__internalHttpRequestHandler__ function instead if it is more convenient for you.

BojanJurca, 2021-03-02:
Please download latest webSererver.hpp file and replace the one you have.

closed 2021-03-02 by BojanJurca

tomoyan596, 2021-03-02:
Hello 😃
Thank you for your prompt reply.

As you said, you don't have to access the URL directly in your browser, so add it to each webServer.hpp if needed 😊

Thank you very much 🙇‍♂️