BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #9: ESP32 stream function is working conditionally with different browser

xiaolaba, 2021-03-02:
hi, ESP32 stream function is working conditionally with different browser
Firefox is not working, Win10 Edge ok, Android chrome ok, any idea whether server side issue or the browser variance ?

here is screen shot with win10 machine, firefox used was ver 86. 32 bit.
[image]
[image]

BojanJurca, 2021-03-02:
Please download the latest webServer.hpp file and replace the one you already have. Firefox should work now.

closed 2021-03-02 by BojanJurca

xiaolaba, 2021-03-04:
head up and salute.
yes, it works.