BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #10: after a while LED control was rendered malfunction

xiaolaba, 2021-03-04:
Testing perhaps 30 minutes or an hour or so, the LED control was no longer functional, unless reboot the ESP32. Any idea for debugging this?. It is latest screen capture of following and remain no control to LED. thanks.
[image]

BojanJurca, 2021-03-04:
Hi. My first guess would be that it is networking problem - ESP32 has limited networking capabilities (if there are too many simultaneous connections). My experience so far is that ESP32 works better in STA mode then in AP mode.  I'll do some testing myself and let you know.  Meanwhile you can check if httpRequest (for LED switch) arrives to ESP32 by adding the following line to httpRequestHandler:

String httpRequestHandler (String& httpRequest, httpServer::wwwSessionParameters *wsp) { // - has to be reentrant!
              // debug:
              Serial.print (httpRequest);

BojanJurca, 2021-03-04:
I've been running this page for more than 3 hours from 3 browsers simultaneously, the same way as you do - through AP, and was unable to replicate the problem. Any suggestions would be useful ...

[image]

xiaolaba, 2021-03-05:
User reply that STA setup with static IP used, and playing those example demo a few, such thing was happening, let us check with users and reproduce such stuff, it is possible one of example html caused such, AP / STA / INTERNET mode all the same. will let you know when I see. thanks.

xiaolaba, 2021-03-05:
hi, this is able to reproduce such user experience with a video recording showed to me.

open index.html, play LED few times, it is good,
go to oscilloscope.html, play Do Digital read, it is ok,
go back to index.html, play LED few times, it is good,
go to oscilloscope.html, play Do Analog read, it is good,
go back to index.html, or any LED control page, play LED again, it is being malfunction,

I could repeat such a few times and confirmed such user feedback, only ESP32 broad itself & 5V power supply, fallback to STA DHCP mode, no any peripheral connected. Perhaps a bug really or coincidently the user just played too hard or too far, but at least it is clear to see that possible causes. In case you will have idea for the clue or how to fix, welcome to share with us. thanks.

And off this topic, NTP was not working when STA with static IP used, no easy for practical server and LAN / WAN. mDNS could be a workaround but it is not port forwarding capability.

BojanJurca, 2021-03-05:
I believe the cause of the problem is that analogRead has was performed (by oscilloscope) ob pin 2 (built-in LED) which corrupted pinMode setting hence consequently called digitalWrite (on pin 2) failed. This sequence of operations is not supported by ESP32. Normally one would have pins for digital input, pins digital output (which you can also read as digital input), pins for analog input and pins for analog output. It all depends on what you intend to connect to those pins. You are not supposed to do analog reading on pin that is supposed to perform digital output.

Try analogRead (from oscilloscope) on pin 32 for example. This should be OK.

BojanJurca, 2021-03-05:
For example, suppose this is your project:
- you want to control built-in LED brightness with PWM
- you want to read analog signal (battery voltage or whatever) connected to pin 32
Than you would configure (probably in setup ()) your pins similar to this:

              // example: PWM on built-in led (GPIO 2)
              ledcSetup (0, 83, 10);                    // channel 0, 83 Hz, 10 bit resolution (1024 values)
              ledcAttachPin (2, 0);                     // attach pin 2 (built-in led) to channel 0
              ledcWrite (0, 307);                       // PWM of built-in LED (channel 0 -> pin 2) with 1/3 of period (307 out of 1023)
              // example: ADC on GPIO 32
              analogReadResolution (12);                // set 12 bit analog read resolution (defaule)
              analogSetPinAttenuation (32, ADC_11db);   // set attenuation for pin 32 to 2600 mV range (default)

If you want to see what is going on in your ESP32 you can configure oscilloscope to perform digitalReads on pin 2 or analogReads on pin 32.

BojanJurca, 2021-03-05:
I'll check NTP with static IP and let you know. Personally I don't use static IPs since it is more convenient and behaviour is more predictable if you configure your router's DHCP to always assign the same IP to your ESP32. But nevertheless NTP should work if ESP32 uses static IP that is not in any contradiction with other IPs that router assigns to its STAtions. 

I'd appreciate if you could check if network connectivity is otherwise OK, for example by telneting to your ESP32 (through AP or your WiFi router) and then try to ping your router from there? Or ping your NTP server (for example ping 1.si.pool.ntp.org)?

I'll perform some tests myself and let you know.

BojanJurca, 2021-03-05:
You are right, the cause for NTP not working lies in static IP set by ESP32. 

If you configure your WiFi router's DHCP to always assign the same IP to your ESP32 and then configure ESP32 to dynamically acquire its IP from DHCP then NTP works as it should. 

If you reconfigure ESP32 to use the same IP in a static manner then NTP stops working.

Honestly, I do not know why is it so. It could be that ESP32_web_ftp_telnet_server_template initialises WiFi in the wrong way when static IP is used or it could be the matter of underlying software. I'll leave this issue open while looking for a solution.

Meanwhile I suggest the use of DHCP reservations on the WiFi router.

BojanJurca, 2021-03-05:
It turned out that it is that ESP32_web_ftp_telnet_server_template initialises WiFi in the wrong way when static IP is used. It seems that it should also configure DNS servers but this would require some additional work and additional entry into /network/interfaces. It will take me a day or two. Beside workaround mentioned above there is  now an obvious second workaround. Use IP instead of NTP server names in /etc/ntp.conf file or in ntpdate -u telnet command.

BojanJurca, 2021-03-05:
Solved. Please download the latest network.h and replace with it the one you have. Then add additional two fields to /network/interfaces which should look something like this:

# WiFi STA(tion) configuration - reboot for changes to take effect
# get IP address from DHCP
#   iface STA inet dhcp

# use static IP address (example below)
iface STA inet static
     address 10.0.0.44
     netmask 255.255.255.0
     gateway 10.0.0.1
     dns1 193.189.160.13
     dns2 193.189.160.23

closed 2021-03-05 by BojanJurca

xiaolaba, 2021-03-07:
> Solved. Please download the latest network.h and replace with it the one you have. Then add additional two fields to /network/interfaces which should look something like this:
>
> '''
> # WiFi STA(tion) [...]

hi, would you please upload the network.h for that patch ,thanks.

BojanJurca, 2021-03-07:
I'm sorry, I thought I did. It is available now.

BojanJurca, 2021-03-07:
BTW you may also want to download latest telnetServer.hpp, there are some putty bug fixes there.

xiaolaba, 2021-03-08:
> 
> 
> I'm sorry, I thought I did. It is available now.

> 
> 
> BTW you may also want to download latest telnetServer.hpp, there are some putty bug fixes there.

excellent, it works very well. thanks