BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #12: use SD card instead of flash

jiangotto, 2021-04-19:
The flash is too small,only 4MB.I'm wondering if we can use SD card filesystem,but i can't make it.Do you have any plans to support SD filesystem. :)

BojanJurca, 2021-04-19:
I'll keep it in mind ...

jiangotto, 2021-04-19:
> I'll keep it in mind ...

THANKS SO MUCH.
I've tried to edit parts of your code but there are something wrong with my code.I will keep debugging.

BojanJurca, 2021-10-12:
If you want to make use of __webServerHomeDirectory__ as well - it is obtained from /etc/passwd as a home directory of system user named webserver. You can change this simply by editing /etc/passwd with built in telnet vi editor.

BojanJurca, 2021-10-13:
I'm closing this issue.

closed 2021-10-13 by BojanJurca

BojanJurca, 2023-08-12:
I kept it in my mind for quite some time, as promised. Now it is possible to use a SD card instead of  or together with a built-in flash disk. Preferred option would be "instead of" since the file system would use less memory for itself this way.  

When you start a httpServer you can specify the server's root directory if it is different from /var/www/html.

jiangotto, 2023-08-13:
Excellent work!
Thanks for your continual efforts on this project.