BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #15: Compile with ESP-IDF

abi55, 2022-07-01:
Can you please guide me to compile this project with ESP-IDF without Arduino component? I am using ESP-IDF v4.4.1.

BojanJurca, 2022-07-01:
Hello. I would really like to help you but I've never used ESP-IDF. Please, share your experience once you succeed.

BojanJurca, 2022-07-01:
If you have any specific question though, I woul like to help if I can.

abi55, 2022-07-04:
Hello, Thank you for quick response. 
Your code is elegant. But somehow IDF compiler does not understand it and throws a lot of compiler errors like fall through if-statements, format-overflow, return-type. I tried to fix all, and compiled using Arduino as a ESP-IDF component. It compiled ok, but after flashing to device I am getting continuous crashes. I guess its Arduino component issue. Also this Arduino component is bulky. For the time being I will pause this exploration and come back when I get some other esp component library to compile your code.

BojanJurca, 2022-07-13:
I have replaced Arduino Strings with std::Strings and delays with esp_task_wdt_reset and vTaskDelay (see the attachment). Haven't tested it thoroughly, though.
I believe that with some extra effort you should be able to port the code to ESP IDF without including Arduino libraries now
[Esp32_web_ftp_telnet_server_template.zip](https://github.com/BojanJurca/Esp32_web_ftp_telnet_server_template/files/9106014/Esp32_web_ftp_telnet_server_template.zip)
.

abi55, 2022-07-18:
I tried compiling this code with ESP-IDF but there are still a lot of dependencies on Arduino core such as timing and filesystem. This is a good start. I will try to replace those with IDF natives. Thanks for your support.

closed 2022-07-18 by abi55