BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #17: W5500 Ethernet Support

aminsteve, 2022-12-12:
Thanks for your great library. I wanted to use ESP32 Ethernet using W5500 chip instead of WiFi, but I had no chance. Even after implementing the W5500 with lwip, the socket function fails and reboots the esp32 again and again. Is there any chance to help me with that?

BojanJurca, 2022-12-12:
Thanky, you are welcome.

Unfortunately I have no experience with Ethernet and W5500. I guess if sockets would have worked with W5500 then the servers would also work, since they use sockets in the way that is very common for implementation of client-server communication. If you can get sockets working on W5500 you shouldn't have any problems.

Can you identify exactly where your ESP32 reboots? Are you sure it is a problem related to sockets?

BojanJurca, 2022-12-14:
... and perhaps if you can send me an output to Arduino console window ...

aminsteve, 2022-12-16:
Finally, I could add ethernet. Now W5500 Ethernet and WiFi are working together.

Here is the ethernet.h file:
```
#include "driver/spi_master.h"

#define CONFIG_EXAMPLE_ETH_SPI_HOST SPI2_HOST

#if defined(ARDUINO_ARCH_ESP32S3)
#pragma message "ESP32S3"
#define CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO 12
#define CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO 11
#define CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO 13
#define CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ 36
#define CONFIG_EXAMPLE_ETH_SPI_CS_GPIO 10
#define CONFIG_EXAMPLE_ETH_SPI_INT_GPIO 3
#else
#pragma message "ESP32"
#define CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO 14
#define CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO 13
#define CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO 12
#define CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ 12
#define CONFIG_EXAMPLE_ETH_SPI_CS_GPIO 15
#define CONFIG_EXAMPLE_ETH_SPI_INT_GPIO 4
#endif
#define CONFIG_EXAMPLE_ETH_SPI_PHY_RST_GPIO -1
#define CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR 1

uint8_t mac_address[6] = {0x02, 0x00, 0x00, 0x12, 0x34, 0x56};

typedef struct
{
    uint8_t spi_cs_gpio;
    uint8_t int_gpio;
    int8_t phy_reset_gpio;
    uint8_t phy_addr;
} spi_eth_module_config_t;

void startEthernet()
{
    ESP_ERROR_CHECK(esp_netif_init());
    
    // Create default event loop that running in background
    // TODO: start event loop if has not been started yet
    // ESP_ERROR_CHECK(esp_event_loop_create_default());

    // Create instance(s) of esp-netif for SPI Ethernet(s)
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_ETH();
    esp_netif_config_t cfg_spi = {
        .base = &esp_netif_config,
        .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH};
    esp_netif_t *eth_netif_spi = NULL;
    char if_key_str[10];
    char if_desc_str[10];
    char num_str[3];

    itoa(0, num_str, 10);
    strcat(strcpy(if_key_str, "ETH_SPI_"), num_str);
    strcat(strcpy(if_desc_str, "eth"), num_str);
    esp_netif_config.if_key = if_key_str;
    esp_netif_config.if_desc = if_desc_str;
    esp_netif_config.route_prio = 30;
    eth_netif_spi = esp_netif_new(&cfg_spi);

    // Init MAC and PHY configs to default
    eth_mac_config_t mac_config_spi = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config_spi = ETH_PHY_DEFAULT_CONFIG();

    // Install GPIO ISR handler to be able to service SPI Eth modlues interrupts
    gpio_install_isr_service(0);

    // Init SPI bus
    spi_device_handle_t spi_handle = NULL;
    spi_bus_config_t buscfg = {
        .mosi_io_num = CONFIG_EXAMPLE_ETH_SPI_MOSI_GPIO,
        .miso_io_num = CONFIG_EXAMPLE_ETH_SPI_MISO_GPIO,
        .sclk_io_num = CONFIG_EXAMPLE_ETH_SPI_SCLK_GPIO,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
    ESP_ERROR_CHECK(spi_bus_initialize(CONFIG_EXAMPLE_ETH_SPI_HOST, &buscfg, SPI_DMA_CH_AUTO));

    // Init specific SPI Ethernet module configuration from Kconfig (CS GPIO, Interrupt GPIO, etc.)
    spi_eth_module_config_t spi_eth_module_config;

    // INIT_SPI_ETH_MODULE_CONFIG
    spi_eth_module_config.spi_cs_gpio = CONFIG_EXAMPLE_ETH_SPI_CS_GPIO;
    spi_eth_module_config.int_gpio = CONFIG_EXAMPLE_ETH_SPI_INT_GPIO;
    spi_eth_module_config.phy_reset_gpio = CONFIG_EXAMPLE_ETH_SPI_PHY_RST_GPIO;
    spi_eth_module_config.phy_addr = CONFIG_EXAMPLE_ETH_SPI_PHY_ADDR;

    // Configure SPI interface and Ethernet driver for specific SPI module
    esp_eth_mac_t *mac_spi;
    esp_eth_phy_t *phy_spi;
    esp_eth_handle_t eth_handle_spi = NULL;
    spi_device_interface_config_t devcfg = {
        .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
        .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
        .mode = 0,
        .clock_speed_hz = CONFIG_EXAMPLE_ETH_SPI_CLOCK_MHZ * 1000 * 1000,
        .queue_size = 20};

    // Set SPI module Chip Select GPIO
    devcfg.spics_io_num = spi_eth_module_config.spi_cs_gpio;

    ESP_ERROR_CHECK(spi_bus_add_device(CONFIG_EXAMPLE_ETH_SPI_HOST, &devcfg, &spi_handle));
    // w5500 ethernet driver is based on spi driver
    eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);

    // Set remaining GPIO numbers and configuration used by the SPI module
    w5500_config.int_gpio_num = spi_eth_module_config.int_gpio;
    phy_config_spi.phy_addr = spi_eth_module_config.phy_addr;
    phy_config_spi.reset_gpio_num = spi_eth_module_config.phy_reset_gpio;

    mac_spi = esp_eth_mac_new_w5500(&w5500_config, &mac_config_spi);
    phy_spi = esp_eth_phy_new_w5500(&phy_config_spi);

    esp_eth_config_t eth_config_spi = ETH_DEFAULT_CONFIG(mac_spi, phy_spi);
    ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config_spi, &eth_handle_spi));

    /* The SPI Ethernet module might not have a burned factory MAC address, we cat to set it manually.
    02:00:00 is a Locally Administered OUI range so should not be used except when testing on a LAN under your control.
    */
    void *mac_address_pointer = mac_address;
    ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle_spi, ETH_CMD_S_MAC_ADDR, mac_address_pointer));

    // attach Ethernet driver to TCP/IP stack
    ESP_ERROR_CHECK(esp_netif_attach(eth_netif_spi, esp_eth_new_netif_glue(eth_handle_spi)));
    
    // TODO: Set static IP from /network/interfaces
    /*
    ESP_ERROR_CHECK(esp_netif_dhcpc_stop(eth_netif_spi));
    char* ip= "192.168.1.95";
    char* gateway = "192.168.1.1";
    char* netmask = "255.255.255.0";
    esp_netif_ip_info_t info_t;
    memset(&info_t, 0, sizeof(esp_netif_ip_info_t));
    
    inet_aton ((const char *)ip, &info_t.ip.addr);
    inet_aton ((const char *)gateway, &info_t.gw.addr);
    inet_aton ((const char *)netmask, &info_t.netmask.addr);
    ESP_ERROR_CHECK(esp_netif_set_ip_info(eth_netif_spi, &info_t));
    */
        
    ESP_ERROR_CHECK(esp_eth_start(eth_handle_spi));
    ESP_ERROR_CHECK(esp_netif_set_hostname(eth_netif_spi, ETHHOSTNAME));
}
```
One just has to include this file and startEthernet after startWifi. There is a lot to do like adding the ethernet to the /network/interfaces file and so on.

Thank you again

BojanJurca, 2022-12-16:
Congratulations. 

I'm glad you have made it. I have just ordered a W5500 myself and would like to try it with your code once it arrives.

Bojan

closed 2022-12-16 by BojanJurca

aminsteve, 2022-12-16:
Oh, great. This way we can add more network management to your code, and there would be no need to restart the ESP after changing the config.