BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino #19: Independent libs

vortigont, 2024-07-16:
This is a nice project indeed. But wouldn't it be more practical to split it into independent libs, i.e. an ftp server, telnet server, etc...
Having an all-in-one is a good demo, but using only a certain components could be difficult in this case.
Cheers!

BojanJurca, 2024-07-16:
Thank you for your suggestion.

The libraries actually are pretty much independent. Taking a look into into moreExamples folder you can find examples such as

A minimal HTTP server.ino,
A minimal Telnet server and client,
...

I'm not 100% sure though that they still compile right away since I've made quite some changes since they have been published.

The other think that you can do is to comment out the libraries that you don't need in Esp32_servers_config.h

If you have any questions I'll be glad to answer.

vortigont, 2024-07-18:
tnaks! Actually I was more interested in FTP server that is not dependent on `loop()` hooks I came by to your implementation. A very nice one I would say.
But there are some "buts" :))
 - it brings dependency on `time_functions.h` which is not that nice (sorry)
 - it brings deps on `/etc/passwd`, quite nice per ce but might not be desirable as a part of some other project
 - definitions in `.h` files are not that good in projects where multiple inclusions are possible, i.e. it brings linker errors on multiple definitions
 - `char __ntpServer1__ [255] = DEFAULT_NTP_SERVER_1;` that is not nice at all, 3x255 bytes of memory persistently allocated

BTW, ESP32 core already contains asynchronous NTP client implemented in lwip lib, any specific reason you ended up with your own implementation?

Thanks!

BojanJurca, 2024-07-18:
Hi,

I can absolutely agree with you that the servers are not perfect. My affection for free code comes from the fact that everyone can see and check how it works and alo modify it if he wants.

Let's see what we can do with the FTP server.

**Dependency on time_functions.h** 

The only reason for bringing time into FTP server is that file creation time is recorded when a file is created and passed (as local time) to FTP client when requested. I guess this dependency could also be removed. I'll check when I get back home.

**Dependency on /etc/passwd**

Normally the FTP server requires you to log in. userManegement.hpp offers 3  possibilities: NO_USER_MANAGEMENT, HARDCODED_USER_MANAGEMENT and UNIX_LIKE_USER_MANAGEMENT - this is the default one and uses /etc/passwd. You can choose another option if you want.

**definitions in .h files**

What would a better way be?

**char __ntpServer1__ [255] = DEFAULT_NTP_SERVER_1; that is not nice at all, 3x255 bytes of memory persistently allocated**

Well, you have to store the server names somewhere in the memory. One way to get around it is to decrease the lengths if you already know which servers you are going to address. In general, server names can consist of 254 characters. Personally I don't like to use F(String(... since Strings do not detect errors when running out of memory which may be quite OK for small sketches but not for the servers.

**any specific reason you ended up with your own implementation (of NTP client)**

Not really. I wanted to log all network traffic for statistics measurement, but NTP really doesn't cause much of the traffic anyway. But the dependency on  time_functions.h can easily be omitted.

BojanJurca, 2024-07-18:
You were right, time_functions.h is not needed. I still recommend setting the current time somehow differently in order to keep file creation time accurate. 

I simplified the code a little bit and prepared A_minimalFTP_server (included here). 
[Minimal_FTP_server.zip](https://github.com/user-attachments/files/16285713/Minimal_FTP_server.zip)

vortigont, 2024-07-24:
I guess some work could be done to segregate things a bit :) will try help with PRs when have some free minute.
NTP is crucial for proper file timestamping, but better leave it out of ftp's scope.

plain `.h` files where function definitions are placed could be really painful if used in multiple .`cpp`'s, like if include time functions in ftp-related cpp file and also in http-related cpp file, linker will yell with errors.
Arduino's ino file is not a real cpps, so it work with examples. But if building a lib - should be better to move function/vars definitions to `cpp` files or hide inside classes.

With `char ntpServer1 [255]` maybe a class with  Arduino's `String` or `std::string` would be better options. Those objects are using some mem anyway but far less then 3x255.

BojanJurca, 2024-07-28:
I made libraries less dependent and also included your changes to time_functions.h. Using existing NTP libraries did not release any memory, though.

Please see the latest Minimal-FTP-server.ino.

closed 2024-07-28 by BojanJurca